jywarren/spectral-workbench #256: Gain-Correction

Stoft1, 2015-02-25:
After some experimentation ( http://publiclab.org/notes/stoft/02-25-2015/plab-spectrometer-gain-correction ) it looks possible that a default gain-correction curve could be included as part of SWB at least as an option. While there are significant limits to the PLab devices from the intensity vs. wavelength calibration, as a true calibration, a first-order approximation seems like a major improvement in the direction of providing users with more meaningful spectral results. As an adjunct to this, it might be helpful to add graphical markers at the wavelength limits -- i.e. ~400nm and ~700nm as beyond these limits the noise takes over as the camera has virtually no relative sensitivity.

jywarren, 2015-03-16:
Hi! Can you please copy this over to our new tracker? This tracker is closed and no longer used. Thanks!

https://github.com/publiclab/spectral-workbench/issues

jywarren, 2015-03-23:
Moved to publiclab/spectral-workbench#107

closed 2015-03-23 by jywarren