jywarren/spectral-workbench #257: Calibrated Wavelength Data

Stoft1, 2015-02-25:
I did two sets of measurments with PLab3.0 and, therefore, two separate calibrations. While the # of data points (and # of wavelength values) is the same, the wavelength values themselves are different ! OOPS! Yes, I'll admit they are close (just a few nm shifted) but for comparing supposidly calibrated spectra (wihtin SWB's present constructs) comparing two, calibrated (different cals) will have a residual offset error between the spectra. This was easily found from the downloaded 'raw' CVS files. There is no reason not to select a pre-defined set of calibration wavelengths for the data -- unless the horizontal resolution changes. IE. the displayed and downloaded data should not have a wavelength offset to the data sets -- but yes, two separate calibrations might have the CFL peaks at slightly different nm values..... but comparing the two data sets should not contain an 'X' axis shift to the data points.

jywarren, 2015-03-16:
Hi! Can you please copy this over to our new tracker? This tracker is closed and no longer used. Thanks!

https://github.com/publiclab/spectral-workbench/issues

jywarren, 2015-03-23:
Moved to publiclab/spectral-workbench#108

closed 2015-03-23 by jywarren