jywarren/spectral-workbench #259: Calibration Points

Stoft1, 2015-02-25:
The calibration procedure is presently based only on the CFL Blue and Green peaks but because of distortions in the typical detected peak shape, it could be better to include the Red. Also, the procedure instructs the user to find the "highest" green peak. However, the Green peak is actually a double-peak and the highest is not always the "same" peak for all devices -- again, due to peak distortions and general device instability. The instructions should have the user pick the "right-most" peak -- ie. the one on the "red" side of the spectrum with the longer wavelength.

jywarren, 2015-03-16:
Hi! Can you please copy this over to our new tracker? This tracker is closed and no longer used. Thanks!

https://github.com/publiclab/spectral-workbench/issues

jywarren, 2015-03-23:
Moved to publiclab/spectral-workbench#110

closed 2015-03-23 by jywarren