jywarren/spectral-workbench #260: Set Curve Downloads

Stoft1, 2015-02-25:
There is a selection of file types for a single curve plot data download but that option is not available for a displayed 'set' of curves. If it is, but is just hard to find, then it should be made available in the same way and same visibility on the page.

jywarren, 2015-03-05:
Oh no, that's why I couldn't find these -- Dave, would you mind closing all of your issues in jywarren/spectral-workbench and re-filing them in https://github.com/publiclab/spectral-workbench/issues ?

Sorry for the confusion -- we just moved everything over, and didn't want to close down the old tracker because we'd lose all the old discussions.

jywarren, 2015-03-23:
Moved to publiclab/spectral-workbench#111

closed 2015-03-23 by jywarren