jywarren/spectral-workbench #261: A fix for the CFL calibration

Stoft1, 2015-03-02:
I discovered why there's been some residual peak misalignment with listed wavelengths after a SWB CFL calibration; which some have attempted to fix using complex polynomial curve-fit algorithms. The present calibration is based only on a linear 2-point approximation. However, it would be better to use a calculation for dispersion based on the diffraction angle light being intercepted by a flat image plane. See research note:  http://publiclab.org/notes/stoft/03-02-2015/cfl-cal-error-explained for more discussion on the topic.

jywarren, 2015-03-16:
Hi! Can you please copy this over to our new tracker? This tracker is closed and no longer used. Thanks!

https://github.com/publiclab/spectral-workbench/issues

jywarren, 2015-03-23:
Moved to publiclab/spectral-workbench#112

closed 2015-03-23 by jywarren