jywarren/spectral-workbench #262: CSV Data Downloads

stoltch, 2015-03-13:
I am probably missing something but when I select CSV box I go to another screen and no indication of  file info to download. I tried all options for data preferences on the boxes and it was the same thing. Is there a problem downloading data right now? Note: this worked before and I was able to download the data in .csv format then something changed. I tried logging in using Chrome and then Explorer with no change.

Thanks,
Chad

jywarren, 2015-03-16:
Hi! Can you please copy this over to our new tracker? This tracker is closed and no longer used. Thanks!

https://github.com/publiclab/spectral-workbench/issues

jywarren, 2015-03-23:
I believe this was fixed in publiclab/spectral-workbench#114

closed 2015-03-23 by jywarren