jywarren/spectral-workbench #263: Custom Calibration not available

rennne, 2015-10-23:
As it is stated in the link, it should somehow exist
http://publiclab.org/wiki/spectral-workbench-calibration#Custom+calibrations

but if I click on that link, it does not work.
Since I have Laser-pointers with a know wavelenght, and no CFL available, I would need that option

jywarren, 2015-11-06:
Hi, sorry if you have done this, but just to check, you have to substitute `SPECTRUM_ID` with the ID # of your spectrum. Did you?

If so, and it didn't work, what did it say when you tried, and with which spectrum did you try it (so we can try to reproduce?)

jywarren, 2016-01-20:
Closing this - if you re-encounter it please file it at https://github.com/publiclab/spectral-workbench/issues/new, thank you!

closed 2016-01-20 by jywarren